mirobot/mirobot-arduino #12: Make sensors availabe on CmdProcessor

schimmi2, 2015-01-13:
May You please add the commands to read the collision-sensors and the line sensors ?
This would allow much additional functionality ( For example in Blockly )

Thank You very much

bjpirt, 2015-01-14:
Yes, I've been thinking about how to best do this. There are a few options:
 - read the state of the sensors with a command, which probably makes sense to do anyway
 - emit messages when e.g. the collision sensors fire or the line follower is off track
I might start with the first and then look at the second. What do you think?

bjpirt, 2015-02-05:
This should be done now:
Commit 0f7b705cc584b908a0cfec5267b393c095604bba implements commands to read the current state of the sensors
Commit dfcf4a0d5b0dc656c2fb6f6068ddd626069a5622 lets you subscribe to changes which will be pushed out whenever the state changes

I've been working on implementing Mirobot support for Snap! (http://snap.berkeley.edu/) which will let you access this functionality:

[image]

Should be released shortly.

closed 2015-02-05 by bjpirt